flipperdevices/Flipper-Android-App #4: Use cross-platform tools?

enurseitov, 2020-08-17:
There's currently a number of tools allowing building apps for different platforms from one code base. I personally have a good experience with
* React Native
* Ionic

From both of these frameworks you can interface peripherals (e.g BLE) without a problem, unless you're doing something very non-standard

LionZXY, 2020-08-17:
This will increase the complexity of the application and will not add benefits.
And answer by vopash:
[image]

photovirus, 2020-08-18:
At this early stage, it seems like iOS, iPadOS and watchOS should be designed around iOS Shortcuts, a feature that Android totally lacks, and Android app can be made modular, which doesn’t work at all with Apple App Store. 

The architectural difference won't go well with crossplatform frameworks, and not adapting to respective platforms just means we get worse apps on both. 

Then there are accessibility issues: crossplatform notoriously sucks at that, and native iOS gets this for free (probably, Android too). And, actually, I've yet to see _any_ crossplatform app (aside from helloworlds), which provide great UX.

closed 2020-08-26 by LionZXY