flipperdevices/Flipper-Android-App #46: [ENHANCEMENT] Add application on F-Droid

RoiArthurB, 2021-04-23:
Hello,

In the scope of releasing the application in a full open-source environment, is it plan to release the app on the F-Droid alternative store ? https://f-droid.org/

LionZXY, 2021-05-03:
Look cool

RoiArthurB, 2021-05-03:
If you need some help, I think @IzzySoft can help you :)

IzzySoft, 2021-05-03:
Quick hints:

* establish [Fastlane structures](https://gitlab.com/snippets/1895688) here in your repo (minimum: `short_description.txt` & `full_description.txt` for `en-US`; strongly recommended: also screenshots; recommended: also icon + featureGraphic; nice: also per-release changelogs and multiple languages)
* optionally ask me to cross-check the structure
* have a new tag with Fastlane inside
* open a [RFP](https://gitlab.com/fdroid/rfp/issues) (Request For Packaging) with F-Droid
* we pick up from there :smiley:

hornetfighter515, 2021-11-09:
I came here to suggest this! Would really like to see this feature or contribute if I can.

LionZXY, 2022-06-17:
Can I sign the f-droid APK with my signature?

IzzySoft, 2022-06-17:
Not directly. But what you're probably looking for is a dual-signed APK (carrying yours **and** F-Droids signature). Look for [reproducible builds](https://f-droid.org/docs/Reproducible_Builds/).

BoBeR182, 2022-08-17:
Any blockers that I can help with or things that need to still be completed?

LionZXY, 2022-08-17:
> Any blockers that I can help with or things that need to still be completed?

It just takes time. If anyone wants to do this, I'll help!

BoBeR182, 2022-09-01:
I have opened a PR to start the metadata needed, you will still need to release this app with a new tag, and hopefully add changelog and other stuff like donation info and website to the RFP at FDroid.

BoBeR182, 2022-09-01:
Some issues where flagged here, https://gitlab.com/fdroid/rfp/-/issues/2187 mainly insecure Gradle configuration.

BoBeR182, 2022-09-01:
Can we reopen this until it's included?

BoBeR182, 2022-09-05:
https://gitlab.com/fdroid/fdroiddata/-/merge_requests/11685  for tracking and help.

Flash1232, 2022-10-22:
How can help get this through?

IzzySoft, 2022-10-22:
I guess it's up to @BoBeR182 currently to accept the suggested changes in the MR.

BoBeR182, 2022-10-25:
Merged the changed and bumped the version to 1.2.2 (599) same one as on the Google Play store.
Looks like there is still some errors with Gradle build process, see the following job for all the errors https://gitlab.com/BoBeR182/fdroiddata/-/jobs/3221672445

IzzySoft, 2022-10-25:
There are a lot of scanner errors in that log. Examples (bottom-up):

```
ERROR: Found Java JAR file at build-logic/plugins/convention/build/libs/convention.jar
ERROR: Found Java compiled class at build-logic/plugins/convention/build/classes/kotlin/main/com/flipperdevices/buildlogic/ApkConfig.class
```

I guess someone accidentally checked in build artifacts. In that case, I'd suggest to remove them again (ideally pruning, but for the build deleting would suffice). If those structures are there on purpose, the question is whether the F-Droid build would succeed if they were not there – in which case a `scandelete` or `rm` entry could be made in the build block, here for `build-logic/plugins/convention/build`. That would then avoid those errors, and the build would continue; it aborted due to the scanner ERRORs:

```
ERROR: Could not build app com.flipperdevices.app: Can't build due to 1128 errors while scanning
```

BoBeR182, 2022-10-25:
@IzzySoft any ideas on this one, https://gitlab.com/BoBeR182/fdroiddata/-/jobs/3221854075
You last recommendation helped.

IzzySoft, 2022-10-25:
Yeah, looks much better already – and the build actually started :smiley:

> Gradle build daemon disappeared unexpectedly (it may have been killed or may have crashed)

I'd say retry. If the error remains, being no Android dev I've no clue, sorry.

LionZXY, 2022-10-26:
It's very strange that it takes 40 minutes to assemble. Even on free githab runners, a clean project takes about 10 minutes to build.

Have you tried disabling the demon?
What command is executed on the server? It now builds everything (even the wearos version). Can you tell it to only build the application (`:instances:app:`)?

I also think i should attach a link to GitHub Workflow: https://github.com/flipperdevices/Flipper-Android-App/blob/dev/.github/workflows/internal.yml#L54

LionZXY, 2022-10-26:
I would also like to say a big thank you to @BoBeR182 for doing this. And to other contributors

BoBeR182, 2022-10-26:
Seems the compiled APK won't install on my phone, can you help debug?
https://gitlab.com/fdroid/fdroiddata/uploads/2e1c4f5ae9c82e1a917a0bb38d66406e/com.flipperdevices.app_599.apk.gz 
https://gitlab.com/fdroid/fdroiddata/-/merge_requests/11685#note_1149898055

IzzySoft, 2022-10-26:
@BoBeR182 have you signed it? The CI spits out an unsigned APK.

BoBeR182, 2022-10-26:
I assumed a rooted phone can just install unsigned APKs, I will learn how to self sign and try again.

IzzySoft, 2022-10-27:
> I assumed a rooted phone can just install unsigned APKs

To my experience not. Most of my devices are rooted, and a simple `adb install` using an unsigned APK always failed (complaining of missing certs).

BoBeR182, 2022-11-16:
Are reproducible builds possible? Or should we move ahead here? This is above my experience level.
https://gitlab.com/fdroid/fdroiddata/-/merge_requests/11685#note_1171890546

IzzySoft, 2022-11-16:
Reproducible builds are not a requirement – while definitely something good to aim for. If it's above your experience, feel free to skip it for now and postpone it (especially if it would keep you from "going live" for an indefinite time). You can always add that later when you feel ready.

BoBeR182, 2022-11-20:
https://gitlab.com/fdroid/fdroiddata/-/merge_requests/11685 is merged, I think we can close this, it should be showing up in the repos soon. There might be work needed to integrate updated changelogs, or other metadata.

LionZXY, 2022-11-22:
Why is there no app icon in f-droid?

IzzySoft, 2022-11-22:
There is no icon in the fastlane structure here (`metadata/en-US/images/icon.png` is missing). And the app is using XML icons only it seems (no PNG), so no PNG there to be used.

TL;DR: can easily be fixed by providing the `icon.png` in `metadata` :wink:

RoiArthurB, 2022-11-23:
Nice, the app is finally available on the F-Droid store (with the app icon for me) ! 🥳🥳

But the application crash when I try to open it for some reason... :/ @LionZXY Do you want to keep tracking that wrong build here or in another issue ? 🤔

LionZXY, 2022-11-23:
It's funny - it turns out no one has tested the version. No DSN for Sentry, I will add to upstream now

LionZXY, 2022-11-23:
```
Process: com.flipperdevices.app, PID: 23835
                                                java.lang.RuntimeException: Unable to get provider io.sentry.android.core.SentryInitProvider: java.lang.IllegalArgumentException: DSN is required. Use empty string to disable SDK.
                                                	at android.app.ActivityThread.installProvider(ActivityThread.java:7493)
                                                	at android.app.ActivityThread.installContentProviders(ActivityThread.java:6999)
                                                	at android.app.ActivityThread.handleBindApplication(ActivityThread.java:6770)
                                                	at android.app.ActivityThread.-$$Nest$mhandleBindApplication(Unknown Source:0)
                                                	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2134)
                                                	at android.os.Handler.dispatchMessage(Handler.java:106)
                                                	at android.os.Looper.loopOnce(Looper.java:201)
                                                	at android.os.Looper.loop(Looper.java:288)
                                                	at android.app.ActivityThread.main(ActivityThread.java:7898)
                                                	at java.lang.reflect.Method.invoke(Native Method)
                                                	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548)
                                                	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:936)
                                                Caused by: java.lang.IllegalArgumentException: DSN is required. Use empty string to disable SDK.
                                                	at ko.n1.d(Sentry.java:824)
                                                	at ko.n1.c(Sentry.java:49)
                                                	at io.sentry.android.core.SentryInitProvider.onCreate(SentryInitProvider.java:113)
                                                	at android.content.ContentProvider.attachInfo(ContentProvider.java:2451)
                                                	at android.content.ContentProvider.attachInfo(ContentProvider.java:2421)
                                                	at io.sentry.android.core.SentryInitProvider.attachInfo(SentryInitProvider.java:15)
                                                	at android.app.ActivityThread.installProvider(ActivityThread.java:7488)
                                                	at android.app.ActivityThread.installContentProviders(ActivityThread.java:6999) 
                                                	at android.app.ActivityThread.handleBindApplication(ActivityThread.java:6770) 
                                                	at android.app.ActivityThread.-$$Nest$mhandleBindApplication(Unknown Source:0) 
                                                	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2134) 
                                                	at android.os.Handler.dispatchMessage(Handler.java:106) 
                                                	at android.os.Looper.loopOnce(Looper.java:201) 
                                                	at android.os.Looper.loop(Looper.java:288) 
                                                	at android.app.ActivityThread.main(ActivityThread.java:7898) 
                                                	at java.lang.reflect.Method.invoke(Native Method) 
                                                	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548) 
                                                	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:936) 
```

LionZXY, 2022-11-23:
What do I have to do to update the version in the F-Droid?
@IzzySoft

IzzySoft, 2022-11-23:
@LionZXY we usually set up apps for auto-update. So tagging a new release should be sufficient. Should the update not turn up a few days later, check back with us. Even better, a few hours after tagging follow the [build monitor](https://monitor.f-droid.org/builds/running), where it eventually should show up.

mrusme, 2022-12-03:
Happy to see this app on F-Droid! However, is there a chance to get rid of this?

```
This app tracks and reports your activity
```

IzzySoft, 2022-12-03:
By removing Sentry (or making it strictly opt-in) I guess? See the above linked MR.

LionZXY, 2022-12-10:
I am going to add toggle for disabling analytics for FDroid users

IzzySoft, 2022-12-10:
Great, thanks! To have the anti-feature removed it must be gone entirely or default to "off", though (see above: removing, or strictly opt-in).

PR #349 (2022-09-01): FDroid metadata
**Background**

To close #46 and add this app to the official fdroid repos.

**Changes**

Added metadata

**Test plan**

Make sure data is accurate
commit e98540c7 (2022-09-01): FDroid metadata (#349)

**Background**

To close #46 and add this app to the official fdroid repos.

**Changes**

Added metadata

**Test plan**

Make sure data is accurate
PR #450 (2022-12-30): Update README.md with download links
Hi,

This small PR adds two buttons to get your app on F-Droid and Google Play, and a link to get your app from the Releases Section 👍 

You can now eventually add the [fdroid](https://github.com/topics/fdroid) and/or [f-droid](https://github.com/topics/f-droid) tags that will appear on the right side of the repo.

close #46
closed 2022-12-30 by LionZXY