flipperdevices/Flipper-Android-App #49: Implement high-level library for communication with Flipper

IlyaGulya, 2021-05-03:
We need a design doc for this library (libraries?).

Main requirements:

- Should be served as a separate library (libraries?)
- Should keep the connection to the device and notify about the connection status.
- Should be implemented using plain listeners (no rxjava/coroutines)
- Should provide Bluetooth device connection flow. (including UI)
- Should provide high-level API to communicate with Flipper core
- Should provide high-level API to communicate with Flipper plugins
- Should provide rxjava/coroutines adapters

LionZXY, 2021-05-03:
> Should provide high-level API to communicate with Flipper plugins

I think in MVPv1 we don't need this

LionZXY, 2021-05-03:
> Should provide high-level API to communicate with Flipper core
We don't want implement any logic inside this library. This library response only about stable connection

IlyaGulya, 2021-05-03:
> > Should provide high-level API to communicate with Flipper plugins
> 
> I think in MVPv1 we don't need this

Yes. I've added it to keep in mind when we will be thinking about the architecture of library.
This features can be separate modules, for sure.

IlyaGulya, 2021-05-16:
# DRAFT
So, here's the design proposal.

We will have two separate features.
1. **Flipper BT communication kit** (further - **btcomm**)
2. **Flipper BT connection UI kit** (further - **btui**): 

### BTComm
This module will give access to low-level communication between the application and Flipper.
Here's draft for this module API:
```
/**
 * Entrypoint into communication with flipper over Bluetooth
 */
interface FlipperConnectionManager {

    /**
     *
     */
    fun scan(callback: ScannerCallback): Session

    fun getPairedDevices(callback: PairedDevicesCallback)

    fun pairWith(mac: MacAddress, callback: PairingCallback): Session

    fun connect(mac: MacAddress)

    fun disconnect(mac: MacAddress)

    fun addDeviceStatusListener(listener: DeviceStatusListener)

    fun removeDeviceStatusListener(listener: DeviceStatusListener)

    fun addIncomingMessageListener(listener: IncomingMessageListener)

    fun removeIncomingMessageListener(listener: IncomingMessageListener)

    fun sendMessage(mac: MacAddress, message: RawFlipperMessage)

    fun interface IncomingMessageListener {

        fun onNewMessage(message: RawFlipperMessage)

    }

    fun interface DeviceStatusListener {

        fun onDeviceStatusChanged(device: FlipperDevice, status: ConnectionStatus)

    }

    interface PairingCallback {

        fun onPairingSuccess()

        fun onPairingFailed(error: Throwable)

    }

    interface PairedDevicesCallback {

        fun onPairedDevicesReady(devices: List<FlipperDevice>)

    }

    interface ScannerCallback {

        fun onFlipperDevicesFound(devices: List<FlipperDevice>)

    }

    interface Session {

        fun release()

    }

    inline class MacAddress(
        val mac: String
    )

    data class RawFlipperMessage(
        val type: Int,
        val data: ByteArray,
    )

    data class FlipperDevice(
        val name: String,
        val mac: MacAddress,
        val paired: Boolean,
    )

    enum class ConnectionStatus {
        DISCONNECTED,
        CONNECTING,
        CONNECTED,
    }
}
```

### BTUI
This module will provide UI for device management (connection/removal/error resolving).

closed 2021-08-31 by LionZXY