flipperdevices/Flipper-Android-App #175: Flipper zero soft bricks when screen streaming

Minarchistshanks, 2022-03-15:
**Describe the bug**
When streaming the screen and using the remote buttons on the phone if you scroll too fast or switch between menus too fast the flipper will freeze and you have to reset by holding left and back.

**To Reproduce**
Steps to reproduce the behavior:
1. Start streaming flipper screen
2. Jump quickly between menus or scroll quickly.

**Additional context**
Issue could be on flipper firmware end. I have latest official firmware. Seems to vary in how long it takes to crash. I haven't attempted going slower with scrolling to see if it's about the speed that I'm scrolling through the flipper or happens no matter what speed I control flipper.

LionZXY, 2022-03-15:
Thanks for the report! I'll look into it.

LionZXY, 2022-03-17:
This is bug from firmware bug. Thank you for creating bug in firmware
https://github.com/flipperdevices/flipperzero-firmware/issues/899

closed 2022-03-17 by LionZXY