flipperdevices/Flipper-Android-App #197: Flipper App tells me "Unsupported Firmware" in spite of me being on release channel.

LordPsycho202, 2022-04-04:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect Flipper Zero to App
2. On first Tap, recieve notice to update Flipper Zero on PC
3. Plug Flipper Zero into qFlipper, which shows Flipper Zero to be on the latest official Released version

**Environment**
Flipper Zero: firmware 0.53.0
App Version: 1.0 (Google Playstore Version)
*Phone*
 Samsung Note 20 Ultra (Android 12)
Patch version March 1, 2022

LionZXY, 2022-04-05:
Does this happen even after restarting the application?

factoryhero, 2022-04-05:
Try unpairing your Flipper in the Bluetooth settings of your phone and then pair it again in the App.
I was experiencing same issue, this one helped.

LordPsycho202, 2022-04-05:
unpairing and repairing seems to have resolved the "unsupported firmware" bug, so it's definitely a workaround.  Now the app is crashing when I try to start a sync, but that's why we test.

LionZXY, 2022-04-05:
> I was experiencing same issue, this one helped.

This is not okay, i am going to reproduce and fix this bug

LionZXY, 2022-04-05:
> Now the app is crashing when I try to start a sync

Now the app is crashing when I try to start a sync.
O_o
What? Why? Just open application and start synchronization?

LordPsycho202, 2022-04-05:
First point: Agree.  The disconnecting/reconnecting might be a valid work around, but it doesn't excuse the initial bug.

Second point: Whenever I do this, the App immediately disappears.  This is whether I start it from the main screen or from the debug menu.

LionZXY, 2022-04-11:
Fixed this in latest dev

closed 2022-04-11 by LionZXY