flipperdevices/Flipper-Android-App #209: Sync crashes

qqmajikpp, 2022-04-23:
**Describe the bug**
A clear and concise description of what the bug is.

Samsung S20+
Android 12

Syncing worked the first time on my device. Now syncing just causes the app the crash. Tried clearing data and cache. Remove and reinstall does not change issue. File manager and streaming work fine.

**To Reproduce**
Steps to reproduce the behavior:
1. Switch on...
2. Press button '....'
3. Wait for the moon phase
4. It burns

**Expected behavior**
A clear and concise description of what you expected to happen.

**Logs**
Add debug logs

**Target**
Specify the target

**Additional context**
Add any other context about the problem here.

LionZXY, 2022-04-23:
What version do you have?

MVGavrilov, 2022-05-28:
Same problem with samsung a8, the app crashes in few (3-4) sec after connection to the flipper. FW version: 4b4576b, same problem was with one of the previous versions (8... dont remember). App version: 1.1.1

LionZXY, 2022-05-28:
Try latest beta version, please

MVGavrilov, 2022-05-28:
1.1.2.320 (pre release) same problem

LionZXY, 2022-05-28:
Use the latest release in the official Flipper Zero firmware please.  
Repeatable?

MVGavrilov, 2022-05-28:
Same with release 0.58.1

LionZXY, 2022-05-28:
Please, remove `Link (Link�s Awakening).nfc` file

closed 2022-06-27 by LionZXY