flipperdevices/Flipper-Android-App #218: Samsung "my files" file manager bug

SHMAUS-Carter, 2022-05-08:
Using the Samsung's my files I am unable to upload any file to flipper

**Expected behavior**
 Flipper shows up in open with list.
For supported files types

**Target**
1.0

**Additional context**
Works correctly with any third party file manager such as FX or a deep link file manager that calls the native android "files" file manager.

components/deeplink/impl/src/main/AndroidManifest.xml

LionZXY, 2022-05-13:
Is there somewhere I can download this file manager?

GottZ, 2022-08-06:
same issue with https://play.google.com/store/apps/details?id=pl.solidexplorer2

GottZ, 2022-08-06:
fx doesn't work for me either.
file is a .txt

Programistich, 2022-08-06:
Hi, I can try reproduce your issue and try fix this
Thx for report!

LionZXY, 2022-08-10:
Fixed in https://github.com/flipperdevices/Flipper-Android-App/commit/2ee18a3fcc7e25bfb7d4478062002150576233ec for import from file manager

closed 2022-08-10 by LionZXY