flipperdevices/Flipper-Android-App #234: updater: Local firmware build detected as "Release", unlike qFlipper

digitalcircuit, 2022-06-09:
**Describe the bug**
The Flipper Android app assumes `FirmwareChannel.RELEASE` if it is not able to parse the firmware version and incorrectly offers to install an update.  This contrasts with qFlipper, which assumes `CUSTOM` if it fails to parse the version.

**To Reproduce**
Steps to reproduce the behavior:
1. Build the Flipper firmware locally in a branch **not** named `dev`, `rc`, or `release`
2. Install your local firmware build
3. Connect to the Flipper Android app
4. Observe the `Firmware Version` and `Firmware Update` sections

**Expected behavior**
The Flipper Android app correctly detects the use of a local firmware build and does not attempt to offer an update.

However, the Flipper Android app should still provide a way to change the `Update Channel` back to `Release`, `Release Candidate`, or `Development` to provide a way to migrate away from a local firmware build.

**Logs**
*Unsure of which logs would be relevant - a full `adb logcat`…?*

**Target**
Pixel 4 XL running Android 12 (SP2A.220505.002)

**Additional context**
The Flipper Android app [builds the firmware version from a string in `buildFirmwareVersionFromString()` from `FirmwareVersionBuildHelper.kt`](https://github.com/flipperdevices/Flipper-Android-App/blob/d20ef4bb82ea71486b7b3727712e05eb8fb9cc2a/components/updater/impl/src/main/java/com/flipperdevices/updater/impl/utils/FirmwareVersionBuildHelper.kt#L14-L45 ).  This falls through to `FirmwareChannel.RELEASE`.

qFlipper [builds the firmware version from a string in `branchToChannelName()` from `deviceinfohelper.cpp`](https://github.com/flipperdevices/qFlipper/blob/73c2ac6ac119cfaf5cabbc5c24f7cfc1a3f4b582/backend/flipperzero/helper/deviceinfohelper.cpp#L262-L283 ).  This falls through to `CUSTOM`.

To fix this in the Android app in a way that allows reverting to a release/RC/dev build, this may involve adding a new firmware channel that can't be manually selected, but the app selects it when it can't parse the version.

digitalcircuit, 2022-06-10:
It's more of a visual bug - the app claims there's an update available (big, green `Update` button), when instead it should act as if switching update channel, offering to replace the firmware (orange `Install` button).

## qFlipper behavior
[image]

qFlipper shows firmware version as `unknown` and sets the update button to `Install`, signifying it will replace the existing unknown firmware.

## Android app behavior
### Current - treats as `RELEASE`
[image]

The Android app acts as if the Flipper is running release firmware and offers to install an update, not making it clear that the "update" will actually be reverting the local firmware build to the official release.

The version parsing is also a bit weird - it's using my local Git branch name (`local-region`) as if it was a version number.

### Expected - treats as custom with option to replace
[image]
> Firmware on Flipper doesn't match the selected update channel.  Selected version will be installed.

This is close to the behavior I'd expect, though the version parsing is still messed up.  It'd be nice if the Flipper would distinguish the local firmware version like `Dev [commit hash]`, but in a different color, perhaps `Custom [commit hash]` or such.

LionZXY, 2022-07-18:
Fixed by https://github.com/flipperdevices/Flipper-Android-App/commit/5ed8ffd70e9cb9128be817b380bb7e14e209fbb5

digitalcircuit, 2022-07-18:
I've updated to the latest Android app dev build and can confirm this works.

Thank you!