flipperdevices/Flipper-Android-App #284: Recursive file indexing in the "Archive" tab

jkctech, 2022-07-20:
In the archive tab in the app, there is no way to list files that are sorted in folders, only files directly in the subfolders are listed.

**Describe the solution you'd like**
A folder like "browser" or all files indexed recursively and listed as a single list would be handy to have.

Foul, 2022-08-08:
And I'll add : listed alphabetically if possible.

LionZXY, 2022-08-08:
> And I'll add : listed alphabetically if possible.

Then the order of the keys will be different on Flipper and on Android.

evaneaston, 2022-12-27:
@jkctech do you mean allow viewing irs and subs  that 
are in folders below the /ext/infrared and /ext/subghz?  If so, I completely agree.  Otherwise we have to resort to longer file names to categorize by device.  And then the names are to long to show in a widget or favorite on the flipper.

Not-swasd, 2023-08-03:
is there any update?

jdrch, 2024-07-06:
> > And I'll add : listed alphabetically if possible.
> 
> Then the order of the keys will be different on Flipper and on Android.

Perhaps, but for those of us with many keys the *status quo* makes finding them time consuming. This in turns reduces the Flipper Zero's usefulness as the delay makes it quicker to just use the original 1st party key.