flipperdevices/Flipper-Android-App #324: Wrong date clock

Foul, 2022-08-08:
### Describe the bug.

When linked/synced to the PC APP, date clock is ok.
When linked/synced to the Android APP, date clock is wrong.
[image]

### Reproduction

Switch on your flipper zero
Sync the Android APP via Bluetooth
Check your date/year under Clock App !
(day/month are good)

### Target

Android App.

LionZXY, 2022-08-08:
Please, check again with latest beta

Foul, 2022-08-10:
same thing with beta from today :(

LionZXY, 2022-08-10:
Oops) Thank you!

closed 2022-08-10 by LionZXY

Foul, 2022-08-11:
Seems to work normally with latest fw and app. You can let close.

commit c4efc920 (2022-08-10): Add year to rtc (Fix #324)