flipperdevices/Flipper-Android-App #393: Hide or minimize update button when up to date firmware

kowalski7cc, 2022-09-25:
# What are you want to add or change

Update button should not be so important when flipper is up to date.

# What questions do you have

Users should be allowed to re-update or update from a custom firmware/unknown version
Users should have a custom firmware install button

LionZXY, 2022-09-25:
This will cause jumps in the interface and result in an inconsistent design

kowalski7cc, 2022-09-26:
Maybe something like change button color and text:
[image]

kowalski7cc, 2022-09-27:
Edit: so I updated anyway even if my Flipper was up to date. Flipper crashed and mobapp showed correctly "no updates"

LionZXY, 2022-10-03:
This was updated region, most likely.

closed 2022-10-03 by LionZXY