flipperdevices/Flipper-Android-App #400: crash on start of app

VAKazakov, 2022-10-04:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Start app, pair with flipper or start app with already paired flipper
2. Crash

**Expected behavior**
A clear and concise description of what you expected to happen.
App works

**Logs**
eb965b42f14d4efd8382db45f3f71789 - from app
from serial:
```
145682 [I][BtGap] Connection parameters: Connection Interval: 39 (48 ms), Slave                                                                                                                                Latency: 0, Supervision Timeout: 500
145685 [W][BtGap] Unsupported connection interval. Request connection parameters                                                                                                                                update
146128 [I][BtGap] Connection parameters event complete
146130 [I][BtGap] Connection parameters: Connection Interval: 30 (37 ms), Slave                                                                                                                                Latency: 0, Supervision Timeout: 500
146309 [I][BtGap] Pairing complete
146312 [I][BtSrv] Open RPC connection
146609 [I][BtGap] Connection parameters event complete
146611 [I][BtGap] Connection parameters: Connection Interval: 6 (7 ms), Slave La                                                                                                                               tency: 0, Supervision Timeout: 500
146613 [W][BtGap] Unsupported connection interval. Request connection parameters                                                                                                                                update
146795 [I][BtGap] Connection parameters event complete
146798 [I][BtGap] Connection parameters: Connection Interval: 30 (37 ms), Slave                                                                                                                                Latency: 0, Supervision Timeout: 500
147458 [I][BtGap] Rx MTU size: 414
147758 [I][BtGap] Connection parameters event complete
147761 [I][BtGap] Connection parameters: Connection Interval: 12 (15 ms), Slave                                                                                                                                Latency: 0, Supervision Timeout: 500
147763 [W][BtGap] Unsupported connection interval. Request connection parameters                                                                                                                                update
147923 [I][BtGap] Connection parameters event complete
147926 [I][BtGap] Connection parameters: Connection Interval: 30 (37 ms), Slave                                                                                                                                Latency: 0, Supervision Timeout: 500
153559 [I][BtGap] Disconnect from client. Reason: 13
153563 [I][BtSrv] Close RPC connection
```
**Target**
Specify the target
0.68.1-rc, hardware_ver: 12 (flipper)
1.2.1.559 (app)

**Additional context**
Add any other context about the problem here.

ljorge705, 2022-10-04:
logcat
```java
time: 1664907061010
msg: java.lang.NumberFormatException: For input string: "1,61"
stacktrace: java.lang.NumberFormatException: For input string: "1,61"
	at jdk.internal.math.FloatingDecimal.readJavaFormatString(FloatingDecimal.java:2054)
	at jdk.internal.math.FloatingDecimal.parseDouble(FloatingDecimal.java:110)
	at java.lang.Double.parseDouble(Double.java:543)
	at a2.b.l(PlaceholderExtensions.android.kt:5)
	at an.e.a(LazyJVM.kt:4)
	at a2.f.q(TtsAnnotationExtensions.android.kt:10)
	at bd.h.b(ComposableInfoCardContent.kt:32)
	at bd.h.a(ComposableInfoCardContent.kt:60)
	at bd.h$a.c0(ComposableInfoCardContent.kt:1)
	at f0.i.m0(Composer.kt:44)
	at f0.i$h.z(Composer.kt:20)
	at e.f.H(Unknown Source:8)
	at f0.i.X(Composer.kt:23)
	at f0.i.k0(Composer.kt:4)
	at f0.u.r(Composition.kt:7)
	at f0.p1.r(Recomposer.kt:9)
	at f0.v1$a.i0(Recomposer.kt:63)
	at androidx.compose.ui.platform.e0$c.doFrame(Unknown Source:8)
	at androidx.compose.ui.platform.d0$d.doFrame(AndroidUiDispatcher.android.kt:15)
	at android.view.Choreographer$CallbackRecord.run(Choreographer.java:1229)
	at android.view.Choreographer$CallbackRecord.run(Choreographer.java:1239)
	at android.view.Choreographer.doCallbacks(Choreographer.java:899)
	at android.view.Choreographer.doFrame(Choreographer.java:827)
	at android.view.Choreographer$FrameDisplayEventReceiver.run(Choreographer.java:1214)
	at android.os.Handler.handleCallback(Handler.java:942)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loopOnce(Looper.java:201)
	at android.os.Looper.loop(Looper.java:288)
	at android.app.ActivityThread.main(ActivityThread.java:7893)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:550)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:936)
	Suppressed: yn.p0: [f0.f1@8fefb7f, androidx.compose.ui.platform.h1@86bf74c, d2{Cancelling}@c19a395, d0@1003caa]

```

VAKazakov, 2022-10-04:
Additional information:
after tinkering with hardware, i managed to start app reliably with out installed sd card
at top of the screen will appear notification about FW update not possible until sd card will be installed
it is not going out even after card installement, and hitting syncronise

SD Card field shows Not Found

access to file system on SD is possible

but, when i hit Retry in Firmware update window, it thinks a second or two and crashes. Do ya wanna me reproduce and send you logs?

Programistich, 2022-10-05:
Yeah, thx for report, if fix that in #401

Programistich, 2022-10-05:
Try use next version 1.2.1.568 and give feedback

VAKazakov, 2022-10-05:
Couldn't find 1.2.1.568, so took 1.2.1.569, everything works now, no crash, update successful

closed 2022-10-05 by VAKazakov