flipperdevices/Flipper-Android-App #423: Opening .sub files inside a folder

Llama-Master, 2022-10-22:
I know I have posted about this multiple times, but I need a reply.

Here is what I need, and if you could, please help me

I cannot open a .sub file from my android app if the file is inside a folder. I have them inside a folder to organize them to find them easier. The problem is, I cannot open them while they are inside a folder to broadcast to my flipper from my device. What I am trying to say is please make it possible to open them while they are in a folder, it would make things much easier.

Thank you 
[Mario-head (1).txt](https://github.com/flipperdevices/Flipper-Android-App/files/9845122/Mario-head.1.txt)

whudson, 2022-11-11:
This is a duplicate of https://github.com/flipperdevices/Flipper-Android-App/issues/284

whudson, 2022-11-11:
Seems like a duplicate of https://github.com/flipperdevices/Flipper-Android-App/issues/284

LionZXY, 2022-11-13:
We are going to add it

closed 2022-11-13 by LionZXY