flipperdevices/Flipper-Android-App #446: Кнопка "update" горит зелёным всегда

a-n-d-r, 2022-11-18:
**Описание бага**
Кнопка "update" горит зелёным даже в том случае, когда последняя версия прошивки уже установлена

**Как повторить**
1. Установить последнюю версию прошивки на флиппер (канал release)
2. Очистить все данные мобильного приложения
3. Очистить все данные флиппера через qFlipper
4. Подключить флиппер к приложению по bluetooth
5. Кнопка будет гореть зелёным

**Как "пофиксить"**
1. Начать обновление с мобильного приложения
2. На 1-2% прервать его через кнопку "cancel"
3. Кнопка будет гореть серым

**Ожидаемое поведение**
Кнопка горит серым, когда флиппер итак обновлен до последней версии прошивки

**Телефон**
Android 10, Doogee S86

LionZXY, 2022-11-20:
It's a normal behaviour

LionZXY, 2022-11-20:
The fact is that this is more of a region update than firmware in this case.

closed 2022-11-20 by LionZXY