flipperdevices/Flipper-Android-App #447: Allow Sending INFRARED signals

TQMatvey, 2022-11-21:
**Is your feature request related to a problem? Please describe.**
why is it possible to send subghz signal, emulate NFC, RFID, iButton, but not infrared?

**Describe the solution you'd like**
Allow sending Infrared signals from App -> Archive -> Infrared signal in same way as i would send a subghz signal, or any other supported signal

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

evaneaston, 2022-12-27:
I agree this would be nice.

LionZXY, 2022-12-29:
We will add this, in the future, it is already planned

TQMatvey, 2023-09-08:
still waiting :D

akoskoronka, 2023-10-21:
The first thing I noticed when I tried the app remote control was the lack of this feature.
It would be very nice, could use the flipper as an ir blaster.

TQMatvey, 2023-10-22:
> We will add this, in the future, it is already planned

well, it's been almost a year from me opening the issue, still no feature present..

LionZXY, 2024-05-14:
This was done a long time ago, forgot to close it

closed 2024-05-14 by LionZXY