flipperdevices/Flipper-Android-App #461: [Regression] Upgrade on final release do not show up on v1.3.0.689

AkechiShiro, 2022-12-10:
**Describe the bug**
Latest final release 0.73.1 doesn't show up for upgrade

**To Reproduce**
Steps to reproduce the behavior:
1. Switch on BT
2. Go on app
3. Wait for the sync to finish
4. It shows that 0.72.1 is the latest final release even if the latest tagged release is 0.73.1 on the Flipper Zéro firmware repository.

**Expected behavior**
An update should be shown to be made when a new final release tag is out on the official FP zero FW repository.

**Logs**
Add debug logs

**Target**
Specify the target

**Additional context**
The final release tags went from v0.72.1 to v0.73.1 and seems to have skipped v0.73.0.

The v1.2.1 of the app does work properly, and finds the update.

LionZXY, 2022-12-10:
Could you, please, try again?

AkechiShiro, 2022-12-10:
@LionZXY I've upgraded already my firmware, I'll check with the next update.

closed 2022-12-10 by AkechiShiro