flipperdevices/Flipper-Android-App #476: F-Droid build failed

linsui, 2022-12-26:
```
> Task :components:bridge:pbutils:compileReleaseKotlin
w: Argument -Xopt-in is deprecated. Please use -opt-in instead
e: /builds/linsui/fdroiddata/build/com.flipperdevices.app/components/bridge/pbutils/src/main/java/com/flipperdevices/bridge/protobuf/StreamKtx.kt: (5, 27): Unresolved reference: protobuf
e: /builds/linsui/fdroiddata/build/com.flipperdevices.app/components/bridge/pbutils/src/main/java/com/flipperdevices/bridge/protobuf/StreamKtx.kt: (6, 27): Unresolved reference: protobuf
e: /builds/linsui/fdroiddata/build/com.flipperdevices.app/components/bridge/pbutils/src/main/java/com/flipperdevices/bridge/protobuf/StreamKtx.kt: (16, 21): Unresolved reference: MainKt
e: /builds/linsui/fdroiddata/build/com.flipperdevices.app/components/bridge/pbutils/src/main/java/com/flipperdevices/bridge/protobuf/StreamKtx.kt: (17, 5): Not enough information to infer type variable T
e: /builds/linsui/fdroiddata/build/com.flipperdevices.app/components/bridge/pbutils/src/main/java/com/flipperdevices/bridge/protobuf/StreamKtx.kt: (27, 13): Unresolved reference: main
e: /builds/linsui/fdroiddata/build/com.flipperdevices.app/components/bridge/pbutils/src/main/java/com/flipperdevices/bridge/protobuf/StreamKtx.kt: (28, 17): Unresolved reference: hasNext
e: /builds/linsui/fdroiddata/build/com.flipperdevices.app/components/bridge/pbutils/src/main/java/com/flipperdevices/bridge/protobuf/StreamKtx.kt: (37, 13): Unresolved reference: main
e: /builds/linsui/fdroiddata/build/com.flipperdevices.app/components/bridge/pbutils/src/main/java/com/flipperdevices/bridge/protobuf/StreamKtx.kt: (38, 17): Unresolved reference: hasNext
```

https://gitlab.com/linsui/fdroiddata/-/jobs/3523527711

Could you please take a look? Thanks!

LionZXY, 2022-12-26:
Thank you for report!

LionZXY, 2022-12-29:
Try just running it again, I don't see how to reproduce it

closed 2022-12-29 by LionZXY