flipperdevices/Flipper-Android-App #477: Mobile app

KrymsonClouds, 2022-12-26:
### Describe the bug.

Android 13, Google pixel 7 pro app keeps crashing before i can sync with flipper.

### Reproduction

Unknown

skotopes, 2022-12-26:
@LionZXY this one is for you ;-)

LionZXY, 2022-12-28:
Which version of application?

ase1590, 2023-01-03:
Axum here, 
I am able to reproduce this on my newly purchased Pixel 6a.

The app will find and pair successfully with the Flipper, however during what looks to be the sync process, the app will crash. 

App version: 1.4.0.739
Android version: 13 with December security patch
Kernel: 5.10.107-android13-4-00038-ga6f610f1f1e6-ab9083768 #1 Tue Sep 20 02:33:52 UTC 2022

avinal, 2023-01-03:
Able to reproduce it on my Pixel 4a. The bluetooth connects fine and then the app crashes.

ase1590, 2023-01-30:
This issue seems to be fixed now in 1.4.0.783

avinal, 2023-01-30:
Is there a release I can install?

ase1590, 2023-01-30:
> Is there a release I can install/test?

Can grab the latest from the play store, as thats the version I'm using. 
File sync seems to be broken, but at least the app doesnt totally crash out now, and other features can be used.

avinal, 2023-01-30:
The latest on Play Store is 1.4.0 (updated 27 Dec 2022). And that one doesn't work.

ase1590, 2023-01-30:
@avinal can grab the apk from https://github.com/flipperdevices/Flipper-Android-App/releases/download/1.4.0.783/flipper-zero-gms-1.4.0.783.apk and load it yourself

avinal, 2023-01-30:
Thanks @ase1590 can confirm it works as expected ☺️

slartybartfasd, 2023-02-12:
> @avinal can grab the apk from https://github.com/flipperdevices/Flipper-Android-App/releases/download/1.4.0.783/flipper-zero-gms-1.4.0.783.apk and load it yourself

Thanks for your effort! Sadly, v.738 does not work for me. Pixel 6, Android 13, fw: unlshd-027e. App and sync works one time after restarting the phone (which is my only workaround). Deleting cache and data does not fix it. Forget Flipper does not fix it.

ase1590, 2023-03-04:
Version 1.4.1.834 seems to support file sync and all other features. 
I'd recommend re-testing now on this version. If it all works as expected, we can probably vote to have this issue closed.

avinal, 2023-03-05:
@ase1590 I can confirm it works as expected on Pixel 4a, Android 13.

Silveere, 2023-03-12:
I just got my Flipper a day ago and I seem to also be having this issue. I am on the GitHub build of version 1.4.1.838, also on a Pixel 7 Pro, Android 13, and the issue still seems to occur. Rebooting seems to fix it temporarily, but the crashing always seems to come back. For additional context, I am on firmware XFW-0042 ([ClaraCrazy/Flipper-Xtreme](https://github.com/ClaraCrazy/Flipper-Xtreme)), although the issue seems to persist on stock firmware and ever survive a full reset of the Flipper and uninstall/reinstall of the app. From what I've seen in this thread and a few others, this seems to be a Pixel specific issue.

ase1590, 2023-03-12:
I'm not able to reproduce this on the play store version of 1.4.1.838 with my phone

Phone/OS info:
Pixel 6a
stock Android 13 w/ Feb 5 security patch
Build: TQ1A.240205.001.D2

@Silveere are you on stock android? Also have you attempted to try the play store version?

Silveere, 2023-03-12:
My phone is rooted with Magisk, but other than that it is stock. I have tried both the Play Store and GitHub versions, they have the same issue. It doesn't seem to be related to root, though; logcat shows the same exact permission error that you had:
<details><summary>logcat</summary>
<pre><code>03-12 17:13:01.523 30431 30431 I FlipperInformationApi: Battery level is 96
03-12 17:13:01.568  2129  2257 D BluetoothGatt: onConnectionUpdated() - Device=80:E1:27:9D:5B:24 interval=12 latency=0 timeout=500 status=0
03-12 17:13:01.568 30431 30446 D BluetoothGatt: onConnectionUpdated() - Device=80:E1:27:9D:5B:24 interval=12 latency=0 timeout=500 status=0
03-12 17:13:01.573 30431 30431 I FlipperBleManager: From BleManager: Connection parameters updated (interval: 15.0ms, latency: 0, timeout: 5000ms)
03-12 17:13:01.589 30431 30431 I FlipperBleManager: From BleManager: Data written to descr. 00002902-0000-1000-8000-00805f9b34fb, value: (0x) 01-00
03-12 17:13:01.589 30431 30431 I FlipperBleManager: From BleManager: Notifications enabled
03-12 17:13:01.614 30431 30431 I FlipperBleManager: From BleManager: Read Response received from 00002a1a-0000-1000-8000-00805f9b34fb, value: (0x) 6F
03-12 17:13:01.614 30431 30431 D BluetoothGatt: setCharacteristicNotification() - uuid: 00002a1a-0000-1000-8000-00805f9b34fb enable: true
03-12 17:13:01.622 30431 30431 I FlipperInformationApi: Battery power state is 111
03-12 17:13:01.647 30431 30431 I FlipperBleManager: From BleManager: Data written to descr. 00002902-0000-1000-8000-00805f9b34fb, value: (0x) 01-00
03-12 17:13:01.648 30431 30431 I FlipperBleManager: From BleManager: Notifications enabled
03-12 17:13:01.681 30431 30431 I FlipperBleManager: From BleManager: Read Response received from 03f6666d-ae5e-47c8-8e1a-5d873eb5a933, value: (0x) 30-2E-31-34
03-12 17:13:01.681 30431 30431 D BluetoothGatt: setCharacteristicNotification() - uuid: 19ed82ae-ed21-4c9d-4145-228e61fe0000 enable: true
03-12 17:13:01.684 30431 30431 D AndroidRuntime: Shutting down VM
--------- beginning of crash
03-12 17:13:01.684 30431 30431 E AndroidRuntime: FATAL EXCEPTION: main
03-12 17:13:01.684 30431 30431 E AndroidRuntime: Process: com.flipperdevices.app, PID: 30431
03-12 17:13:01.684 30431 30431 E AndroidRuntime: java.lang.SecurityException: Need BLUETOOTH PRIVILEGED permission: Neither user 10598 nor current process has android.permission.BLUETOOTH_PRIVILEGED.
03-12 17:13:01.684 30431 30431 E AndroidRuntime: 	at android.app.ContextImpl.enforce(ContextImpl.java:2240)
03-12 17:13:01.684 30431 30431 E AndroidRuntime: 	at android.app.ContextImpl.enforceCallingOrSelfPermission(ContextImpl.java:2268)
03-12 17:13:01.684 30431 30431 E AndroidRuntime: 	at android.content.ContextWrapper.enforceCallingOrSelfPermission(ContextWrapper.java:948)
03-12 17:13:01.684 30431 30431 E AndroidRuntime: 	at com.android.bluetooth.Utils.enforceBluetoothPrivilegedPermission(Utils.java:408)
03-12 17:13:01.684 30431 30431 E AndroidRuntime: 	at com.android.bluetooth.gatt.GattService.permissionCheck(GattService.java:463)
03-12 17:13:01.684 30431 30431 E AndroidRuntime: 	at com.android.bluetooth.gatt.GattService.registerForNotification(GattService.java:3782)
03-12 17:13:01.684 30431 30431 E AndroidRuntime: 	at com.android.bluetooth.gatt.GattService$BluetoothGattBinder.registerForNotification(GattService.java:1073)
03-12 17:13:01.684 30431 30431 E AndroidRuntime: 	at com.android.bluetooth.gatt.GattService$BluetoothGattBinder.registerForNotification(GattService.java:1061)
03-12 17:13:01.684 30431 30431 E AndroidRuntime: 	at android.bluetooth.IBluetoothGatt$Stub.onTransact(IBluetoothGatt.java:812)
03-12 17:13:01.684 30431 30431 E AndroidRuntime: 	at android.os.Binder.execTransactInternal(Binder.java:1285)
03-12 17:13:01.684 30431 30431 E AndroidRuntime: 	at android.os.Binder.execTransact(Binder.java:1244)
03-12 17:13:01.699  1598  5387 W ActivityTaskManager:   Force finishing activity com.flipperdevices.app/com.flipperdevices.singleactivity.impl.SingleActivity
03-12 17:13:01.713 30431 30431 I Process : Sending signal. PID: 30431 SIG: 9</code></pre>
</details>

Phone Info:
- Pixel 7 Pro
- Android 13, build TQ1A.221205.011
- December 5 Security Update
- February 1 Google Play System Update

tkbstudios, 2023-03-29:
I will try to fix it when I get home, seems that the BLUETOOTH permission for android is missing in the app, gonna add it later.

hitoshi-5963, 2023-06-16:
In order to replace the micro SD card in the flipper zero main unit with a high performance one, after taking a backup with qflipper and the Android mobile app, I installed the new micro shin card, updated it with the normal version, and then tried to install the custom firmware. An error occurred while updating the mobile app.

LionZXY, 2023-10-20:
Let's try again on latest version

closed 2023-10-20 by LionZXY