flipperdevices/Flipper-Android-App #478: Spamming the "play alert on flipper" button in the app freezes the flipper

jqlifice, 2022-12-26:
### Describe the bug.

If you click that button repeatedly while the last alert has not finished yet it will play them all after each other (probably kinda expected) also while these alerts play the flipper does not accept new key inputs and mine froze after finishing playing the alerts. 

### Reproduction

1. click play alert on flipper in app
2. :repeat: a large amount of time (I clicked 30 times)
3. wait for it to finish playing the alerts
4. mine was frozen

this does not happen when only clicking a few times and is also not always reproduceable.

skotopes, 2022-12-26:
I've checked code: normally companion application should wait for call complete confirmation before activating button again.

skotopes, 2022-12-26:
@LionZXY ping me if you need details on this one.

closed 2023-01-26 by jqlifice