flipperdevices/Flipper-Android-App #532: mfkeys app incorrectly displays and saves keys that contain leading zeros

fersingb, 2023-03-01:
**Describe the bug**
Keys that start with a `0` are not displayed/saved properly. The leading zero is missing, resulting in an invalid key

**To Reproduce**
Steps to reproduce the behavior:
1. create a file `.mfkey32.log` with the following content: `Sec 0 key A cuid 2f05ec42 nt0 bebf5b50 nr0 d34d3251 ar0 a4463233 nt1 a3cfc299 nr1 507d9fde ar1 dac188a7`
2. Save if in /ext/nfc/
3. Run the Android mfkey app in NFC Tools
4. Notice that the leading 0 is missing

**Expected behavior**
Leading 0 is correctly displayed and saved

fersingb, 2023-03-23:
@LionZXY you actually reproduced it. The key is missing the first `0` The key should be `0a1b2c3d4e5f`

LionZXY, 2023-03-23:
Oh, thank you, it's my fault

commit 55c5342d (2023-03-24): Fix mfkey32 formatting (Fix #532) (#547)

**Background**

Right now we format number in wrong way

**Changes**

- Use `"%012X".format(key)` instead of `key?.toString(radix =
16)?.uppercase()`

**Test plan**

Find reproducable steps in  #532
closed 2023-03-24 by LionZXY