flipperdevices/Flipper-Android-App #577: Can't build due to implicit dependencies.

iehgit, 2023-04-30:
FAILURE: Build failed with an exception.

* What went wrong:
A problem was found with the configuration of task ':components:nfc:tools:impl:externalNativeBuildRelease' (type 'ExternalNativeBuildTask').
  - Gradle detected a problem with the following location: '/home/hagbard/src/Flipper-Android-App/components/nfc/tools/impl/build/intermediates/cxx/RelWithDebInfo/5tu37e2y'.
    
    Reason: Task ':components:nfc:tools:impl:mergeInternalNativeLibs' uses this output of task ':components:nfc:tools:impl:externalNativeBuildRelease' without declaring an explicit or implicit dependency. This can lead to incorrect results being produced, depending on what order the tasks are executed.
    
    Possible solutions:
      1. Declare task ':components:nfc:tools:impl:externalNativeBuildRelease' as an input of ':components:nfc:tools:impl:mergeInternalNativeLibs'.
      2. Declare an explicit dependency on ':components:nfc:tools:impl:externalNativeBuildRelease' from ':components:nfc:tools:impl:mergeInternalNativeLibs' using Task#dependsOn.
      3. Declare an explicit dependency on ':components:nfc:tools:impl:externalNativeBuildRelease' from ':components:nfc:tools:impl:mergeInternalNativeLibs' using Task#mustRunAfter.
    
    Please refer to https://docs.gradle.org/8.0.2/userguide/validation_problems.html#implicit_dependency for more details about this problem.

LionZXY, 2023-04-30:
Where are the sources from? Everything works fine on CI

iehgit, 2023-04-30:
The sources are cloned directly out of this very repository.

LionZXY, 2023-05-01:
Please investigate your environment, everything works on CI, it is impossible to reproduce the problem

closed 2023-05-01 by LionZXY