flipperdevices/Flipper-Android-App #593: Remove Countly tracker library.

EvanCarroll, 2023-05-18:
Because of the nature of the Flipper Zero, it seems like it should show no tracker libraries. Currently TC registers Countly. It would be nice if there was a build switch to build without this, at least for F-Droid.

The metric gathering is done here,
https://github.com/flipperdevices/Flipper-Android-App/blob/4edd4c6791d2694289133e1bc332880d30183f3d/components/analytics/metric/impl/src/main/java/com/flipperdevices/metric/impl/api/MetricApiImpl.kt#L10

It seems like it's posting to this site, https://countly.lionzxy.ru/ which doesn't seem to exist. I'm guessing this resolves locally for development and gets set in Gradle when building here, 

https://github.com/flipperdevices/Flipper-Android-App/blob/4edd4c6791d2694289133e1bc332880d30183f3d/.github/workflows/release.yml#LL62C1-L63C1

This telemetry is used in the calls to `reportSimpleEvent` and `reportComplexEvent`, which are used in things like the [MiFare screen](https://github.com/flipperdevices/Flipper-Android-App/blob/4edd4c6791d2694289133e1bc332880d30183f3d/components/nfc/mfkey32/screen/src/main/java/com/flipperdevices/nfc/mfkey32/screen/viewmodel/MfKey32ViewModel.kt#LL128C19-L128C36). It sounds unwise and incriminating to have a third party with all the metadata (ip, timestamp) of when they were using RFID functionality of the Flipper.

LionZXY, 2023-05-18:
First of all, it is worth mentioning that all collected information is completely **anonymous** and does **not contain user data**. From the collected information it is **impossible** to recover the user's data or his identity. 
Secondly, it is worth noting that the next build of FDroid will no longer contain metrics, this is the responsibility of the environment variable "ORG_GRADLE_PROJECT_is_metric_enabled". 

You can verify the purity of the metrics, since all of our code is open-source and the builds are reproducible

closed 2023-05-18 by LionZXY

EvanCarroll, 2023-06-03:
@LionZXY F-Droid is still showing telemetry is enabled on the build,

https://f-droid.org/packages/com.flipperdevices.app/

> Anti-Features
This app has features you may not like. [Learn more!](https://f-droid.org/en/docs/Anti-Features/) [This app tracks and reports your activity](https://f-droid.org/en/docs/Anti-Features/#Tracking)

EvanCarroll, 2023-06-03:
@Wjxfi don't be a dick, @LionZXY has quite obviously tried to remove the tracker libraries for just us F-Droid users. It seems like he is trying though not successful. I also want the trackers removed. F-Droid is doing the right thing by saying that there are trackers. And @LionZXY seems to be doing the right thing by acknowledging this is something at least some of his users are very passionate about and trying to scrub them.

The real take away here is that F-Droid is adding tremendous value by at least pointing out that there is potentially incriminating telemetry on the app. Moreover, for the time being you can use TrackerControl to block the telemetry.