flipperdevices/Flipper-Android-App #658: App catalog doesn't work for F-Droid build

dogtopus, 2023-07-26:
**Describe the bug**

The category doesn't work (shows everything regardless of the selection) and installing any app will fail silently (shows progress bar but immediately went back to showing the "Install" button without installing the app). Almost exactly like when it was still in dummy mode recently.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to app catalog
2. Try to install anything or use the category

**Expected behavior**

The category and app installation works as it should.

**Logs**
Add debug logs

**Target**

App: 1.6.1 (fdroid) running on Android 13.

Flipper: 0.87.0 on f7

**Additional context**
Add any other context about the problem here.

LionZXY, 2023-07-26:
Could you, please, attach report I'd from "Report Bug" button in options menu?

dogtopus, 2023-07-26:
@LionZXY Report bug button seems to crash the app on my phone.

I can't access the logcat due to absence of PC but I can provide it later in a few days if you'd like me to. Meanwhile you may be able to reproduce it locally as well.

LionZXY, 2023-08-02:
Fix already merged in dev, so waiting for fdroid to update their version.

LionZXY, 2023-08-03:
https://gitlab.com/fdroid/fdroiddata/-/commit/a7752747adc45eb4fb0e703faac23a83eb1d3f76

dogtopus, 2023-08-04:
1.6.2 is on F-Droid. Unfortunately it doesn't seem to fix the issue.

Logcat output that might be relevant:
```
08-04 10:43:56.517 17537 17579 E FapQueueRunner: While executing Install(applicationAlias=pokemon, applicationUid=64971d4d577d519190ede5c6, applicationName=Pokemon Trading, toVersion=FapItemVersion(id=64971d4e577d519190ede61e, version=1.2, target=Received(target=f7, sdk=34.3), buildState=READY), categoryAlias=GPIO, iconUrl=https://catalog.flipperzero.one/api/v0/application/version/assets/64971d4d577d519190ede5c7)
08-04 10:43:56.517 17537 17579 E FapQueueRunner: ho.d: Client request(GET https://catalog.flipperzero.one/api/v0/application/version/64971d4e577d519190ede61e/build/compatible?=&=) invalid: 400 Bad Request. Text: "{"detail":{"status":"error","details":"You must provide at least one parameter"}}"
```

Also the "report bug" button still crashes the app. The logcat output isn't very useful in this case due to code minification.

LionZXY, 2023-08-07:
Sounds very strange - version 1.6.3 has now been released, once fdroid is updated it should fix it

LionZXY, 2023-08-07:
I checked out the August 5 version from F-Droid. The installation works. Please check again

dogtopus, 2023-08-07:
It works for me as well. Thank you very much for the quick fix.