flipperdevices/Flipper-Android-App #667: Phantom Bluetooth

SuperVDF, 2023-08-01:
/### Describe the bug.

Flipper showing active Bluetooth connection when Bluetooth is not on, on mobile device. Using Android 13 latest update. Screen grab provided below.

### Reproduction

Don't use the flipper for 1-3 days, Bluetooth shows connected.

[image]

skotopes, 2023-08-01:
@LionZXY looks like this belongs here

LionZXY, 2023-08-07:
Looks like expected behavior - please provide more information and steps to reproduce

closed 2023-08-07 by LionZXY

SuperVDF, 2023-08-17:
Hey there, so this issue is most certainly closed now. The phantom Bluetooth connection issue appears to have gone away with the last firmware update. To clarify, the flipper zero would display an active Bluetooth connection (--©) even when Bluetooth on the device (my phone in this case) would be off. This would happen after 2 to 3 days of inactivity with the F0 device. Thank you to the Dev team for looking at this.