flipperdevices/Flipper-Android-App #695: NFC in-app Functionalities

NikoCosmico01, 2023-09-08:
### Describe the enhancement you're suggesting.

When editing HEX dump in-app you cannot see the last bit (out of screen on the right corner).
Implement way to copy/paste whole block.
Suggest to implement "import dump from file" with following formats .mct/.json/.bin/.mdf/.dump/.eml .
Implement way to edit in-app (add or view) user keys (mf_classic_dict_user.nfc).

### Anything else?

Trying to work on it.

skotopes, 2023-09-08:
In that case this issue belongs here.

LionZXY, 2023-10-03:
@NikoCosmico01 please, provide phone model and also try reproduce bug again on newest app version

NikoCosmico01, 2023-10-03:
> @NikoCosmico01 please, provide phone model and also try reproduce bug again on newest app version

Samsung Galaxy S23+. Bug is still present on latest version 1.6.4.1211

bubblec0re, 2024-07-28:
Same here. Samsung Galaxy S21 Ultra, latest app version
[image]
[image]
[image]
. 
Last bit not visible in NFC dump editor, but visible when opened through file explorer.