flipperdevices/Flipper-Android-App #722: App crashes during launch after update to 1.6.5 from f-droid

Frohlix, 2023-10-21:
**Describe the bug**
After the latest update (1.6.5. from f-droid), the app crashes on launch.

To be extra precise, it still shows the splash screen with the logo, then the overview/update page with greyed-out fields for a fraction of a second, and then it crashes.

If I delete the data/reinstall, the same happens: Splash screen, first setup page for a split-second, crash

**To Reproduce**
1. Have an Android 11 device (OxygenOS)
2. Install 1.6.5 from f-droid
3. Launch the app
4. Crash

**Expected behavior**
The app to launch and be usable like normal.

**Logs**
`adb logcat | grep flipper`
```
10-21 10:02:03.081  1939  2903 I ActivityTaskManager: START u0 {act=android.intent.action.MAIN cat=[android.intent.category.LAUNCHER] flg=0x10000000 cmp=com.flipperdevices.app/com.flipperdevices.singleactivity.impl.SingleActivity} from uid 10315 pid 2398
10-21 10:02:03.121  1939  2149 D OpRestartProcessManager: updateSelf :  com.flipperdevices.app, size : 30
10-21 10:02:03.133  1939  2150 D OPFD_CTRL_SVC: [2]handle Starting Window for { com.flipperdevices.app }, Dark? false
10-21 10:02:03.146  1939  2161 I ActivityManager: Start proc 1400:com.flipperdevices.app/u0a405 for pre-top-activity {com.flipperdevices.app/com.flipperdevices.singleactivity.impl.SingleActivity}
10-21 10:02:03.153  1285  7061 E ANDR-RAMBOOST_SERVER: ramboost cmd_len = 126 cmd: iop_start -1 com.flipperdevices.app /data/app/~~KSbLa14gzxAQ3UyVqbVqrg==/com.flipperdevices.app-9Pe4FB0Gn0L9pLBg74kgqw== false
10-21 10:02:03.159  1285  7063 E ANDR-RAMBOOST_SERVER: ramboost cmd_len = 128 cmd: iop_start 1400 com.flipperdevices.app /data/app/~~KSbLa14gzxAQ3UyVqbVqrg==/com.flipperdevices.app-9Pe4FB0Gn0L9pLBg74kgqw== false
10-21 10:02:03.160  1939  2689 D OpColorDisplayService: frontPackageChanged: com.flipperdevices.app
10-21 10:02:03.160  1939  2689 D OpProximityController: noteFrontPackageChanged pkg:com.flipperdevices.app uid:10405 lpkg:com.vincent_falzon.discreetlauncher luid:10315
10-21 10:02:03.161  1939  2689 E OpScreenModeService: getAppToken ActivityRecord{1b79b0c u0 com.flipperdevices.app/com.flipperdevices.singleactivity.impl.SingleActivity t17169}
10-21 10:02:03.161  1939  2689 E OpScreenModeService: setRefreshRate token ActivityRecord{1b79b0c u0 com.flipperdevices.app/com.flipperdevices.singleactivity.impl.SingleActivity t17169} request refresh rate=-1 tmpRate -1
10-21 10:02:03.161  1939  2689 E OpScreenModeService: setRefreshRate token ActivityRecord{1b79b0c u0 com.flipperdevices.app/com.flipperdevices.singleactivity.impl.SingleActivity t17169} app request rateid=1 fps=90.0
10-21 10:02:03.162  1939  2689 D OpQuickReply: setQuickReplyResumed focusedApp ActivityRecord{1b79b0c u0 com.flipperdevices.app/com.flipperdevices.singleactivity.impl.SingleActivity t17169} pkgName com.flipperdevices.app
10-21 10:02:03.162  1939  2689 D Hypnus  : hypnusSetScene: uid=10405, pid=1400, name=com.flipperdevices.app, activity=null
10-21 10:02:03.182  1939  2149 D OpScreenModeService: updateRefreshRate token ActivityRecord{1b79b0c u0 com.flipperdevices.app/com.flipperdevices.singleactivity.impl.SingleActivity t17169} rate=1
10-21 10:02:03.203  1400  1400 I TetheringManager: registerTetheringEventCallback:com.flipperdevices.app
10-21 10:02:03.307  1400  1400 I SingleActivity: Create new activity with hashcode: 205840267 and intent Intent { act=android.intent.action.MAIN cat=[android.intent.category.LAUNCHER] flg=0x10000000 cmp=com.flipperdevices.app/com.flipperdevices.singleactivity.impl.SingleActivity }
10-21 10:02:03.494  1400  1400 I DeepLinkHelper: On new intent: Intent { act=android.intent.action.MAIN cat=[android.intent.category.LAUNCHER] flg=0x10000000 cmp=com.flipperdevices.app/com.flipperdevices.singleactivity.impl.SingleActivity }
10-21 10:02:03.494  1400  1400 I DeepLinkParser: Try parse intent Intent { act=android.intent.action.MAIN cat=[android.intent.category.LAUNCHER] flg=0x10000000 cmp=com.flipperdevices.app/com.flipperdevices.singleactivity.impl.SingleActivity }
10-21 10:02:03.494  1400  1400 W DeepLinkParser: Failed parse intent Intent { act=android.intent.action.MAIN cat=[android.intent.category.LAUNCHER] flg=0x10000000 cmp=com.flipperdevices.app/com.flipperdevices.singleactivity.impl.SingleActivity }
10-21 10:02:03.511  1939  2166 D ArtManagerInternalImpl: /data/misc/iorapd/com.flipperdevices.app/1270/com.flipperdevices.singleactivity.impl.SingleActivity/compiled_traces/compiled_trace.pb doesn't exist
10-21 10:02:03.514  1939  2177 I ===> HoustonPackage: addLaunchTimeInfo: for com.flipperdevices.app/com.flipperdevices.singleactivity.impl.SingleActivity isCold:true launchTime:425
10-21 10:02:03.514  1939  2177 I 322     : com.flipperdevices.app#77#425
10-21 10:02:03.515  1939  2166 I ActivityTaskManager: Displayed com.flipperdevices.app/com.flipperdevices.singleactivity.impl.SingleActivity: +425ms
10-21 10:02:03.652 10334 10334 I PhoneStatusBarView: isFullAndNotchEnabled() isFullMode=false pkg=com.flipperdevices.app isCameraNotchIgnoreSetting=false, uid=10405
10-21 10:02:03.911  1400  1400 I FlipperService-74323529: Start flipper service
10-21 10:02:03.916  1400  1400 I FlipperService-74323529: On bind Intent { cmp=com.flipperdevices.app/com.flipperdevices.bridge.service.impl.FlipperService }
10-21 10:02:03.923  1400  7078 E AndroidRuntime: Process: com.flipperdevices.app, PID: 1400
10-21 10:02:03.929  1939  3153 W ActivityTaskManager:   Force finishing activity com.flipperdevices.app/com.flipperdevices.singleactivity.impl.SingleActivity
10-21 10:02:03.933  1939  7083 D FrameworkEventCollector: recordAppIssue PN: com.flipperdevices.app
10-21 10:02:03.935  5564  5564 I GoogleInputMethodService: GoogleInputMethodService.onStartInput():1894 onStartInput(EditorInfo{EditorInfo{packageName=com.flipperdevices.app, inputType=0, inputTypeString=NULL, enableLearning=false, autoCorrection=false, autoComplete=false, imeOptions=0, privateImeOptions=null, actionName=UNSPECIFIED, actionLabel=null, initialSelStart=-1, initialSelEnd=-1, initialCapsMode=0, label=null, fieldId=-1, fieldName=null, extras=null, hintText=null, hintLocales=[]}}, false)
10-21 10:02:04.046  1939  5139 I WindowManager: WIN DEATH: Window{5955107 u0 com.flipperdevices.app/com.flipperdevices.singleactivity.impl.SingleActivity}
10-21 10:02:04.046  1939  5139 W InputDispatcher: Attempted to unregister already unregistered input channel '5955107 com.flipperdevices.app/com.flipperdevices.singleactivity.impl.SingleActivity (server)'
10-21 10:02:04.047  1939  4109 I ActivityManager: Process com.flipperdevices.app (pid 1400) has died: prcp TOP 
10-21 10:02:04.069  1939  4109 D OpColorDisplayService: lpackageName: com.flipperdevices.app
10-21 10:02:04.070  1939  4109 D OpProximityController: noteFrontPackageChanged pkg:com.vincent_falzon.discreetlauncher uid:10315 lpkg:com.flipperdevices.app luid:10405
10-21 10:02:04.432  1939  2149 W ActivityTaskManager: Activity top resumed state loss timeout for ActivityRecord{1b79b0c u0 com.flipperdevices.app/com.flipperdevices.singleactivity.impl.SingleActivity t-1 f}}
```

**Target**
OnePlus BE2029, aka OnePlus Nord N10 5G, running stock OxygenOS 11

**Additional context**
Un- and reinstalling does not fix the crash
Keeping #719 in mind, yes, Google Play is installed
Switching Bluetooth off does not help

akoskoronka, 2023-10-21:
Same here.
Downgrading back to 1.6.4 is the only way to get it back to working state.

```
type: crash
osVersion: google/sunfish/sunfish:13/TQ3A.230805.001/2023100300:user/release-keys
package: com.flipperdevices.app:1270
process: com.flipperdevices.app
processUptime: 466 + 599 ms
installer: com.android.packageinstaller

gq.f: An operation is not implemented.
	at yj.b.a(Unknown Source:151)
	at uk.g.invokeSuspend(Unknown Source:35)
	at uk.g.invoke(Unknown Source:12)
	at androidx.lifecycle.k0.invokeSuspend(Unknown Source:32)
	at androidx.lifecycle.k0.invoke(Unknown Source:12)
	at uq.z.m3(Unknown Source:4)
	at uq.z.n1(Unknown Source:9)
	at androidx.lifecycle.l0.invokeSuspend(Unknown Source:71)
	at mq.a.resumeWith(Unknown Source:8)
	at kr.l0.run(Unknown Source:107)
	at qr.a.run(Unknown Source:93)
	Suppressed: pr.g: [x1{Cancelling}@6a4bff5, Dispatchers.Default]

```

shaohme, 2023-10-23:
Crashes on my GrapheneOS Pixel too.

```
type: crash
osVersion: google/panther/panther:14/UP1A.231005.007/2023101300:user/release-keys
package: com.flipperdevices.app:1270
process: com.flipperdevices.app
processUptime: 174 + 260 ms
installer: com.android.packageinstaller

gq.f: An operation is not implemented.
	at yj.b.a(Unknown Source:151)
	at uk.g.invokeSuspend(Unknown Source:35)
	at uk.g.invoke(Unknown Source:12)
	at androidx.lifecycle.k0.invokeSuspend(Unknown Source:32)
	at androidx.lifecycle.k0.invoke(Unknown Source:12)
	at uq.z.m3(Unknown Source:4)
	at uq.z.n1(Unknown Source:9)
	at androidx.lifecycle.l0.invokeSuspend(Unknown Source:71)
	at mq.a.resumeWith(Unknown Source:8)
	at kr.l0.run(Unknown Source:107)
	at qr.a.run(Unknown Source:93)
	Suppressed: pr.g: [x1{Cancelling}@1d378c1, Dispatchers.Default]

```
version 1.6.5

com.flipperdevices.app
versionCode 1270

targetSdk 33
minSdk 26

Installed: Aug 28, 2023; 10:47
Updated: Oct 21, 2023; 10:37

joserebelo, 2023-10-27:
I believe this is fixed by #723, but I'm not sure (didn't check) why f-droid installs are being considered an unknown source.

ankek, 2023-10-29:
Can confirm that issue. Why it happened only with F-droid app?

LionZXY, 2023-10-30:
1.6.6 released. Expect a release on the F-Droid side

closed 2023-10-30 by LionZXY