flipperdevices/Flipper-Android-App #741: offline cache app description

nanderer, 2023-11-18:
Hi,

The app works grate when connected to the flipper and the internet but when either connection is down not so.

Please cache all installed apps with full description page so i can read them even offline

Thanks

LionZXY, 2023-12-18:
We are now saving the name and icon of the application. Do you really think it's worth caching the whole app info?

closed 2024-01-16 by LionZXY

nanderer, 2024-01-17:
its what i like most about open source repository systems like APT or F-Droid, so i would be glad if you did...

nothing is nicer then browsing available software offline and the anticipation of trying it out when i get internet again, its a feeling in my mind closely linked to open source software. I also am that type of persons that wasts gigabytes of disc space for an offline copy of wikipedia ;)

So if it is not to much work i would really appreciate the extra work for that goal!
On 18.12.23 15:04, Nikita Kulikov wrote:
>
> We are now saving the name and icon of the application. Do you really think it's worth caching the whole app info?