flipperdevices/Flipper-Android-App #754: Document extent of third party firmware support.

EvanCarroll, 2023-12-24:
Previously, I'm 100% sure Flipper Unleashed worked with the Android App on F-Droid. Now I realize it no longer works with the Android App on F-Droid. This issue is **NOT** to get it to work. It's to clarify the extent by which the applications supports third party firmware. If not at all, it would probably save some time to put it on the README.md, I would be willing to send in a patch for it.

It's very confusing as Flipper Unleashed, is marketed as a more featureful fork that disregards regulation and enables more features. However, it has its own Android App, and it's not clear what that provides to the forked firmware or why the Android App must be forked too https://github.com/DarkFlippers/Unleashed-Flipper-Android-App

hedger, 2023-12-24:
It's up to third-party firmwares to maintain compatibility with official tools, not the other way around.

skotopes, 2023-12-25:
@EvanCarroll clearly you should ask this question in 3rd party firmware channels. On our side we don't provide any support for 3rd party firmware.

closed 2023-12-25 by skotopes

EvanCarroll, 2023-12-25:
Why would I ask third party firmware channels to update your docs. I'm just saying,

> _WARNING:_ The Flipper Zero Android App only supports the official Flipper Zero Android Firmware. Other firmwares available for the Flipper Zero, such as the Flipper Unleashed, are not supported though they may or may not work. Please do not file issues on our tracker pertaining to other firmwares.