flipperdevices/Flipper-Android-App #832: Tip me please

rougsig, 2024-04-21:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

rougsig, 2024-04-21:
/bounty $1

algora-pbc, 2024-04-21:
💎 **rougsig** is offering a **$10** bounty for this issue
👉 Got a pull request resolving this? Claim the bounty by adding `@algora-pbc /claim #832` in the PR body and joining algora.io

closed 2024-04-21 by rougsig