flipperdevices/Flipper-Android-App #847: In-app key read & search [NFC, RFID, SubGHz, iButton]

Tomi2965, 2024-05-06:
**Is your feature request related to a problem? Please describe.**
I am always frustrated, when for example somebody found keys with a token, and I must read UID, then via mobile turn on VPN, then connect to our system, and then, after couple of minutes find owner name. We are using three different entry systems, and over 5000 registered tokens. One entry system works only via remote desktop, which is a bit hardcore on mobile. 

**Describe the solution you'd like**
Best solution: add to Flipper App (in Hub) another function, where user can select/upload files with all wanted tokens and names. After, user select token type, read token on FZ, FZ will return UID to app, and app will find owner's name. File format can be primitive, just like CSV (UID,NAME,IS_ACTIVE[Y/N]). After app can make it's own database from primitive entry files for better performance. 

**Describe alternatives you've considered**
My only alternative is currently to read UID with FZ, then go to my office, and find it in entry system. Or do it just in terrain with active VPN via remote control of entry system.

**Additional context**
I know, that FZ have limited HW resources, but, our SmartPhones are capable to do complicated tasks, like indexed searching. So I will be seriously happy, when somebody can bring this idea to life. It's important to not mix this "tokens database" with currently saved tokens in archive or on FZ. 

Thanks and have a nice day!

LionZXY, 2024-05-07:
This looks like a specific task for a specific business. I think you can fork the app and add this functionality yourself, I don't think it will be useful to the rest of the Flipper Zero app users

LionZXY, 2024-05-07:
Reopen the issue if you think it's wrong, please.

closed 2024-05-07 by LionZXY

Tomi2965, 2024-05-07:
@LionZXY thanks for reply. To be honest, wrong or right, it depends on angle of view. I understand and accept your arguments. 

It's a bit time ago, that I asked for new search function in FZ-FW. Precisely - after token read, show a name (on FZ LCD), when same token is found. Somebody from Flipper Team tell me, that this function it's maybe planned to new HW version, because actual FZ have a limited HW resources. Of course - I understand. 

My idea is a 90% of my previous function request, and 10% of new. With this idea, is possible to search for same tokens (show name) after UID read. In app, instead of FZ directly. It's not needed to wait to HW Ver:2.0 to bring this function to life. 

I think, additional function for upload a huge custom database of UIDs and names may help to users. I'm thinking, where are used tokens? In books in libraries, at swimming pool, baths, building entry systems, animals and IDK where else. I'm not sure, if I am the only person, who will use this :) I hope, this will helps, and make FZ again to be closer to peoples and help with everyday problems.

The first crazy thought that occurred to me: you have a lot of small kittens. All with RFID. All pure black. You need to find the right one. Of course, you can use a coloured collars or others. Or, you can grab you'r phone and FZ, in app press "search", scan UID from RFID, and you will immediatelly see a name. Without this function, you must scan RFID, and search for UID somewhere in text notes or in MS Excel :). Or before find the right UID in Excel, and then scan. It not depends on order of tasks.

Secondary example is my case: to read name and maybe active status directly in terrain, due to using different entry systems, and a complicated search in these systems.

Third example: at a swimming pool is used one-day tokens. At morning, you can upload actual DB to phone, and then randomly check guests for token validity. 

Reopen or not. It's up to you. :) Maybe my idea is wrong. IDK. I'll be happy for any type of answer (neg/pos). Thanks for you'r time.