flipperdevices/Flipper-Android-App #867: Bluetooth connection to Samsung S24 Ultra doesn't work

kochpet, 2024-05-25:
Bluetooth connection to Samsung S24 Ultra 
doesn't work

**To Reproduce**
Steps to reproduce the behavior:
1. Switch on Bluetooth
2. Press button Connect in Adroid App

**Expected behavior**
A clear and concise description of what you expected to happen.

**Logs**
Add debug logs

**Target**
Specify the target

**Additional context**
Add any other context about the problem here.
[image]

LionZXY, 2024-05-28:
What if you try to connect from your phone settings? And then in the app.
Can you also attach the logs?
https://docs.flipper.net/mobile-app/bug-report

kochpet, 2024-05-28:
The phone settings connection doesn't work either, there is a PIN request but no connection is established. I have attached the debug log file, the first connection attempt in the log is with the app, the second directly via the phone Bluetooth.
[terminal_content_1716922186759.txt](https://github.com/flipperdevices/Flipper-Android-App/files/15474697/terminal_content_1716922186759.txt)

LionZXY, 2024-06-07:
I need logs from the mobile app side. Please send here the ID Bug Report
https://docs.flipper.net/mobile-app/bug-report#wS2g4

jdrch, 2024-07-06:
WFM on my unlocked S24U.

LionZXY, 2024-07-10:
Please, anyone with the same problem check if it reproduces for you under the following conditions:
- Install the official firmware
- Delete all pairs from the phone and from the flipper
And try again

closed 2024-07-29 by LionZXY