flipperdevices/Flipper-Android-App #871: Bluetooth connection nearly completely not working

ttmx, 2024-06-03:
**Describe the bug**
It seems that after pairing, no information gets transmitted between the Flipper and the phone besides it's firmware version.

**To Reproduce**
Steps to reproduce the behavior:
1. Open mobile app
2. Press "Connect" if not connected yet
3. Press "Synchronize" if not Synching yet

**Expected behavior**
The progress climbs somewhat smoothly from 0% to 100%, and the new data synchronized from the Flipper appears in the app (such as the data in the archive, or installed apps in the hub). 

**Logs**
``
Android App logs: [https://termbin.com/k1ut](https://termbin.com/k1ut) (Too large to fit in the issue)
<details><summary>Flipper Logs</summary>
<p>

<CONNECTED>

              _.-------.._                    -,
          .-"```"--..,,_/ /`-,               -,  \ 
       .:"          /:/  /'\  \     ,_...,  `. |  |
      /       ,----/:/  /`\ _\~`_-"`     _;
     '      / /`"""'\ \ \.~`_-'      ,-"'/ 
    |      | |  0    | | .-'      ,/`  /
   |    ,..\ \     ,.-"`       ,/`    /
  ;    :    `/`""\`           ,/--==,/-----,
  |    `-...|        -.___-Z:_______J...---;
  :         `                           _-'
 _L_  _     ___  ___  ___  ___  ____--"`___  _     ___
| __|| |   |_ _|| _ \| _ \| __|| _ \   / __|| |   |_ _|
| _| | |__  | | |  _/|  _/| _| |   /  | (__ | |__  | |
|_|  |____||___||_|  |_|  |___||_|_\   \___||____||___|

Welcome to Flipper Zero Command Line Interface!
Read Manual https://docs.flipperzero.one

Firmware version: 0.93.0 0.93.0 (e8c6e70a built on 12-10-2023)

>: info device
format.major                  : 3
format.minor                  : 3
hardware.model                : Flipper Zero
hardware.uid                  : 
hardware.otp.ver              : 2
hardware.timestamp            : 1701566911
hardware.ver                  : 12
hardware.target               : 7
hardware.body                 : 9
hardware.connect              : 6
hardware.display              : 2
hardware.color                : 2
hardware.region.builtin       : 4
hardware.region.provisioned   : PT
hardware.name                 : 
firmware.commit.hash          : e8c6e70a
firmware.commit.dirty         : false
firmware.branch.name          : 0.93.0
firmware.branch.num           : 0
firmware.version              : 0.93.0
firmware.build.date           : 12-10-2023
firmware.target               : 7
firmware.api.major            : 39
firmware.api.minor            : 2
firmware.origin.fork          : Official
firmware.origin.git           : https://github.com/flipperdevices/flipperzero-firmware
radio.alive                   : true
radio.mode                    : Stack
radio.fus.major               : 1
radio.fus.minor               : 2
radio.fus.sub                 : 0
radio.fus.sram2b              : 16K
radio.fus.sram2a              : 0K
radio.fus.flash               : 24K
radio.stack.type              : 3
radio.stack.major             : 1
radio.stack.minor             : 17
radio.stack.sub               : 3
radio.stack.branch            : 0
radio.stack.release           : 2
radio.stack.sram2b            : 19K
radio.stack.sram2a            : 14K
radio.stack.sram1             : 0K
radio.stack.flash             : 116K
radio.ble.mac                 : DD145B27E180
enclave.keys.valid            : 10
enclave.valid                 : true
system.debug                  : 1
system.lock                   : 0
system.orient                 : 0
system.sleep.legacy           : 0
system.stealth                : 0
system.heap.track             : 0
system.boot                   : 0
system.locale.time            : 0
system.locale.date            : 0
system.locale.unit            : 0
system.log.level              : 5
protobuf.version.major        : 0
protobuf.version.minor        : 20

>: log
Current log level: debug
Use <log ?> to list available log levels
Press CTRL+C to stop...
31849374 [I][BtGap] Disconnect from client. Reason: 13
31849378 [I][BtSrv] Close RPC connection
31849381 [D][GattChar] Updating RPC status char
31849384 [D][GattChar] Updating Flow control char
31849387 [D][RpcSrv] Session terminated
31855857 [I][BtGap] Connection parameters: Connection Interval: 36 (45 ms), Slave Latency: 0, Supervision Timeout: 500
31856027 [D][BtGap] Slave security initiated
31856297 [I][BtGap] Pairing complete
31856303 [I][BtSrv] Open RPC connection
31856307 [D][GattChar] Updating Flow control char
31856312 [D][RpcSrv] Session started
31856315 [D][GattChar] Updating RPC status char
31856320 [D][GattChar] Updating Battery Level char
31857647 [I][BtGap] Rx MTU size: 414
31858096 [I][BtGap] Connection parameters event complete
31858099 [I][BtGap] Connection parameters: Connection Interval: 12 (15 ms), Slave Latency: 0, Supervision Timeout: 500
31858102 [W][BtGap] Unsupported connection interval. Request connection parameters update
31858165 [D][BtGap] Procedure complete event
31858254 [I][BtGap] Connection parameters event complete
31858257 [I][BtGap] Connection parameters: Connection Interval: 36 (45 ms), Slave Latency: 0, Supervision Timeout: 500
31875501 [I][BtGap] Connection parameters event complete
31875503 [I][BtGap] Connection parameters: Connection Interval: 12 (15 ms), Slave Latency: 0, Supervision Timeout: 500
31875506 [W][BtGap] Unsupported connection interval. Request connection parameters update
31875569 [D][BtGap] Procedure complete event
31875658 [I][BtGap] Connection parameters event complete
31875660 [I][BtGap] Connection parameters: Connection Interval: 36 (45 ms), Slave Latency: 0, Supervision Timeout: 500
31892993 [I][BtGap] Connection parameters event complete
31892996 [I][BtGap] Connection parameters: Connection Interval: 12 (15 ms), Slave Latency: 0, Supervision Timeout: 500
31892998 [W][BtGap] Unsupported connection interval. Request connection parameters update
31893062 [D][BtGap] Procedure complete event
31893151 [I][BtGap] Connection parameters event complete
31893154 [I][BtGap] Connection parameters: Connection Interval: 36 (45 ms), Slave Latency: 0, Supervision Timeout: 500
31907347 [D][GattChar] Updating Battery Level char
31924346 [I][BtGap] Connection parameters event complete
31924348 [I][BtGap] Connection parameters: Connection Interval: 12 (15 ms), Slave Latency: 0, Supervision Timeout: 500
31924351 [W][BtGap] Unsupported connection interval. Request connection parameters update
31924415 [D][BtGap] Procedure complete event
31924504 [I][BtGap] Connection parameters event complete
31924507 [I][BtGap] Connection parameters: Connection Interval: 36 (45 ms), Slave Latency: 0, Supervision Timeout: 500

</p>
</details> 

**Additional context**
You will notice my firmware is in 0.93.0, but the exact same behaviour happens with the latest firmware, therefore I believe it is an Android sided issue.
[image]
[image]
There are keys here, but from before I was unable to sync, no new information seemed to actually be transmitted to my device.

All the permissions were given to the app in Android settings.

skotopes, 2024-06-03:
Please update to latest version 0.102, then try to reproduce issue and attach logs from Flipper and Mobile Phone

skotopes, 2024-06-03:
Also please set log level to debug

ttmx, 2024-06-03:
> Please update to latest version 0.102, then try to reproduce issue and attach logs from Flipper and Mobile Phone

It has the exact same behaviour on 0.102.3 (that is where I took the screenshots). Either way, I'll send the logs.
[Flipper Log](https://github.com/user-attachments/files/15540056/terminal_content_1717450857730.txt)
[App Log](https://github.com/user-attachments/files/15540079/flipperapplog.txt)

> Also please set log level to debug

I believe it is already debug? "Current log level: debug" Am I missing anything?

skotopes, 2024-06-03:
Yep, on firmware side it looks alright. Except there is no RPC session being opened.

LionZXY, 2024-06-07:
> no information gets transmitted between the Flipper and the phone

What kind of information do you expect to receive?

ttmx, 2024-06-07:
I expect to receive the keys I have in the Flipper, be able to access it's files, to check it's installed applications, I expect the "full info" submenu to be populated as well. None of this info was present in the app after multiple attempted syncs. Currently I only have a screenshot of the Hub portion of the app not displaying installed apps. 
 [image]

LionZXY, 2024-06-12:
I see that Flipper Zero on file listing (e.g. /any/subghz) gives `command_status: ERROR` / `command_status_value: 1`

1) Are you sure this is all the logs from the flipper that you have for the session? Can you record a video of it?
2) Can you try using a different sdcard?

ttmx, 2024-06-12:
1. These are all the logs, collected as explained in https://docs.flipper.net/mobile-app/bug-report , using https://googlechromelabs.github.io/serial-terminal/ throughout the whole process, and just exporting the logs. The application logs were collected from the start of the app until the end of the procedure, using logcat.

2. Unfortunately this is the only SD card I have, but I have benchmarked it multiple times, fsck'd, and done full read and write tests, and there have been no issues. It is a 64GB Sandisk MicroSD which I previously used to run a raspberry pi.

LionZXY, 2024-07-29:
Can you record a video? I can't play it back and I don't understand at what point what events are happening. 
From an application point of view, flipper just returns an error on any request